ros2/ros2_tracing #12: Warn if trace directory already exists when using tracing tools/commands

christophebedard, 2022-10-14:
When using the `ros2 trace` command to configure LTTng, there's no warning if the trace directory (that the user is about to write to) already exists. LTTng itself simply appends the new trace data to the existing trace data, which can lead to confusing results.

It would be good if the `ros2 trace` command could check if the trace directory already exists, and, if so, print a warning.

This new warning could also be added to the `Trace` launch action.

christophebedard, 2023-03-22:
Well, I see two options:

1. As mentioned above, emit a warning if the trace directory already exists; or
2. Error out (for `ros2 trace`) or throw an error (for the `Trace` action) if the trace directory already exists, _unless_ the user uses a new `--append`/`append=True` option.

@mjcarroll what do you think?

mjcarroll, 2023-03-22:
I think I would prefer option 2.  As you said, you can end up with some weird discontinuities in the data if you appended without realizing it.  I would prefer for it to error out and allow me to fix it.

Another option would be to auto-append an integer in the case of a conflict, but I don't know if that's too complicated.

christophebedard, 2023-03-22:
Thanks for the feedback.

> Another option would be to auto-append an integer in the case of a conflict, but I don't know if that's too complicated.

Yeah, I don't think that's a desirable default behaviour, since I think the user expects the given session name to be used as the trace directory name, even if the trace directory path is printed. At least that's what I would expect.

Tangentially, `tracetools_trace` can [append a timestamp to the session name](https://github.com/ros2/ros2_tracing/blob/e664e5bbd28b42da6430623fc1bcf644d6b6a449/tracetools_trace/tracetools_trace/tools/path.py#L20), but it's not exposed by `tracetools_trace` itself or by `ros2trace`. It's only used by `tracetools_test`. We could probably expose it through `tracetools_trace`/`ros2trace` (opt-in) separately from this issue.

PR #58 (2023-04-18): Error out if trace already exists unless 'append' option is used
Closes #12

This makes the `ros2 trace` command and the `Trace` launch action fail if the trace already exists, unless the `--append-trace` or `append_trace`/`append-trace` option is used, respectively. Example:

```sh
$ ros2 trace -s 'my-trace'  # Generate trace
UST tracing enabled (28 events)
kernel tracing disabled
context (3 fields)
writing tracing session to: /home/chris/.ros/tracing/my-trace
press enter to start...
press enter to stop...
stopping & destroying tracing session
$ ros2 trace -s 'my-trace'  # Mistakenly append t [... cut]
closed 2023-04-18 by christophebedard