ros2/ros2_tracing #13: Real time data tracing

hs2041, 2022-10-19:
Hi, 
Using ros2_tracing, I am able to generate a ust file which provides me with all the trace calls while the system was running. However, is there a way to get these calls while the system is running in real-time?

christophebedard, 2022-10-24:
Hi! The tracer (LTTng) has features to do this. Take a look at the “live mode” and the “network streaming mode” https://lttng.org/docs/v2.13/#doc-tracing-session-mode. However, note that most of the trace analysis tools related to `ros2_tracing` (namely `tracetools_analysis` and Trace Compass) are not designed to support live analysis, so you might need to adapt them, look for a tool that does that (for example, I think people have worked on integrating trace data into Grafana), or build one yourself using the babeltrace2 library.

hs2041, 2022-10-24:
Hey, thanks. I'll look into it.

closed 2022-10-24 by hs2041