ros2/ros2_tracing #14: Can ros2_tracing trace rclpy?

Binohow, 2022-11-02:
Hello, I want  to know if ros2_tracing support to trace rclpy?  Lttng tools  is too weak for python.

christophebedard, 2022-11-02:
Hi! Unfortunately, `rclpy` is not instrumented; you can only trace `rclcpp`.

> Lttng tools is too weak for python.

What do you mean?

christophebedard, 2022-11-02:
Since this has been requested many times, I've opened an issue as a feature request: #15.

Binohow, 2022-11-03:
> Hi! Unfortunately, `rclpy` is not instrumented; you can only trace `rclcpp`.
> 
> > Lttng tools is too weak for python.
> 
> What do you mean?

Thanks for your answer!  Can we use `Lttng` tracer to implement this feature?  I am very interested in how to implement this function.  I want to participate in this meaning work, but I don't where to start.  Which tracer should I choose first?

> Hello, I want to know if ros2_tracing support to trace rclpy? Lttng tools is too weak for python.

christophebedard, 2022-11-07:
LTTng can trace Python applications: https://lttng.org/docs/v2.13/#doc-python-application

vtran5, 2023-04-04:
Hello, maybe a related question, can ros2_tracing support rclc? If not, is there any guide on how to implement it myself? From the [design document](https://github.com/ros2/ros2_tracing/blob/rolling/doc/design_ros_2.md), I understand that I would need to manually add the tracepoint in rclc following the 6-step guide in the **Adding instrumentation**. Is there anything else that I need to do or considered? Do I need to add all the equivalent TRACEPOINT in rclcpp to rclc in order for it to work properly?

christophebedard, 2023-04-05:
It could, but it would require a bit of work. The proper way would be to design a new set of instrumentation points for `rclc`, since the `rclcpp` instrumentation might not necessarily all apply to `rclc` (and vice versa). Once the instrumentation is defined, you can indeed go through the *Adding instrumentation* section of the `ros2_tracing` design document to actually create the tracepoints.

Like I said, that would be the "proper" way. Also, I'm not sure if those tracepoints should be here in `ros2_tracing`/`tracetools`, since `rclc` is more of a separate client library.

However, you could try to just re-use the existing `rclcpp` tracepoints by adding them to `rclc` as if it was `rclcpp`. I did it a couple years ago with the main `rclcpp` tracepoints, so it might need to be updated, but it can be a good starting point: https://github.com/christophebedard/rclc/commit/090a0cbe636682c0ad314e1a62846f57b7609bb2. Someone also did that for `rclpy`: https://github.com/ros2/ros2_tracing/issues/15#issuecomment-1467823953. This is more of a hack, though, as you can see from my comments in that `rclc` commit. You would probably need to process trace data coming from `rclc` differently compared to trace data coming from `rclcpp`.

christophebedard, 2023-04-06:
Since this issue was mostly just a questions, and since I've opened a standalone issue for the actual feature request (#15), I'll close this.

closed 2023-04-06 by christophebedard