ros2/ros2_tracing #16: Gracefully handle older versions of LTTng

christophebedard, 2022-11-22:
Issue moved from GitLab, see: https://gitlab.com/ros-tracing/ros2_tracing/-/issues/142

christophebedard, 2023-04-08:
@cottsay I ran all tracing tests on RHEL 9 with LTTng 2.12, including some tests I'm adding in #54, and everything passes. Given that LTTng (`liblttng-ust-dev`, `python3-lttng`, `lttng-tools`) is now expected to be installed on Linux, and since the lowest common denominator is version 2.12, which works fine, I don't think we really need to do this anymore.

You said recently that you still wanted this to be improved, though, so I'm just wondering why exactly.

cottsay, 2023-04-09:
> so I'm just wondering why exactly.

I think that it's a valid effort to support as many Linux distributions as is reasonably possible in packages which are considered part of the core of ROS 2, including ones which have previously been supported like RHEL 8. I'm really not asking for much here. Assign the ticket to me if you're not willing to do it yourself.

christophebedard, 2023-04-09:
Sorry, that's not what I meant! I'm all for making this kind of improvement.

> including ones which have previously been supported like RHEL 8

Thanks, you answered my question.

PR #59 (2023-04-12): Allow requiring minimum lttng package version for is_lttng_installed
Closes #16

This changes `is_lttng_installed()` to also check for the `lttng` Python package.

This also adds an optional `minimum_version` parameter to `is_lttng_installed()` to check if the `lttng` Python package version is sufficient. This is then used to ignore some tests which require `lttng.BUFFER_*`, which isn't available until version 2.9. RHEL 8 has LTTng 2.8, so those tests fail. With this change, tests run successfully on RHEL 8, or, rather, the tests that can't run are skipped.
closed 2023-04-12 by christophebedard