ros2/ros2_tracing #29: How to trace non-ROS2 functions within ros2_tracing and view in tracecompass

Seangottarun, 2022-12-05:
I'm trying to run tracing for my ROS2 node. I was able to get the `tracetools_analysis` working and I can get the callback durations. I would also like to get further detail about my node. Could you answer the following questions? Thanks!

1. Does `ros2_tracing` support tracing my own non-ROS2 functions? 
- My ROS2 callbacks run different helper functions and I would like to see how long those functions within the callback take
2. Can I view the trace generated by `ros2_tracing` in a tool like `tracecompass`? 
- I'm hoping to see stuff like the flame graph with my helper functions, but currently `tracecompass` only has a list of events and event density
- I also only see ROS2 specific functions

So far I've just used this launch file. Do I need to add tracepoints within my code or something like that?

```python3
import os
import launch
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import Node
from tracetools_launch.action import Trace

def generate_launch_description():
    """Generate launch description."""
    
    return launch.LaunchDescription(
            [
                Trace(
                    session_name='my_session',
                    events_kernel=['sched_switch'],
                ),
                Node(
                    package="my_package",
                    executable="my_executable",
                    name="some_name",
                )
            ]
        )

```

[image]

https://github.com/christophebedard/ros-world-2021-demo
https://gitlab.com/ros-tracing/ros2_tracing/-/raw/master/doc/2021-10-20_ROS_World_2021_-_Tracing_ROS_2_with_ros2_tracing.pdf
[image]

christophebedard, 2022-12-05:
You can instrument your own code with LTTng and then use the `ros2_tracing` tools to enable the tracepoints (`Trace` action or `ros2 trace` command).

If you want to get a flame graph or flame chart for your functions, I'd recommend using `gcc`'s function instrumentation feature ([`-finstrument-functions`](https://gcc.gnu.org/onlinedocs/gcc-8.4.0/gcc/Instrumentation-Options.html)). This adds function calls at the start and end of the instrumented functions. `gcc` only allows you to instrument _all_ functions and exclude individual functions or headers, so I'd recommend using this `gcc` plugin I've created in order to only instrument the individual functions you want: https://github.com/christophebedard/instrument-attribute-gcc-plugin

You can then use LTTng to trigger tracepoints inside the special functions that `gcc` adds: https://lttng.org/docs/v2.13/#doc-liblttng-ust-cyg-profile. You'll need to enable those special tracepoints (`lttng_ust_cyg_profile*`/`lttng_ust_cyg_profile*:func_entry`) like this: https://gitlab.com/ros-tracing/tracetools_analysis/-/blob/9c853b27fc380d763317ce786f0588007db8f33b/tracetools_analysis/launch/profile.launch.py#L29-L30. You alos need to `LD_PRELOAD` LTTng's `liblttng-ust-cyg-profile.so` library (or `liblttng-ust-cyg-profile-fast.so`), but the `Trace` action or `ros2 trace` command should do that for you.

Then you can import the trace into Trace Compass, and it should be able to display the information as a flame chart/graph, see https://archive.eclipse.org/tracecompass/doc/stable/org.eclipse.tracecompass.doc.user/LTTng-UST-Analyses.html#Flame_Chart_View.

I've written code to use trace data generated from the standard ROS 2 instrumentation in Trace Compass (like [Figures 4 and 5 in this paper](https://arxiv.org/pdf/2204.10208.pdf)). You could then compare the flame charts of your functions to your subscription/timer callbacks, etc., e.g., by looking at the results side by side in Trace Compass. However, this is not all easily available yet. You could try to build Trace Compass yourself if you (really) wanted to, see: https://github.com/christophebedard/ros2-message-flow-analysis#analysis. Eventually you'll be able to download Trace Compass and install this feature, though.

Otherwise, if you create your own instrumentation, then you'll need to do the data analysis yourself (using `tracetools_anlaysis` or by writing a custom Trace Compass plugin). This is why I recommend going the `-finstrument-functions` way for your use-case here.

Hope that's useful. Let me know if you have any questions!

Seangottarun, 2022-12-06:
Thanks for the detailed reply Christophe! Looks really useful.

It will take me some time to familiarize myself with the tools and try this out. I'll let you know if I have any questions, but feel free to close this issue in the meantime.

christophebedard, 2022-12-06:
Sounds good!

closed 2022-12-06 by christophebedard