ros2/ros2_tracing #44: Find solution to allow tracing

christophebedard, 2023-02-14:
As mentioned in #40, we need to start tracing before the application launches to collect initialization information/metadata.

We current workaround is to start tracing 

One possible solution would be to store the metadata and

christophebedard, 2025-06-05:
I had a thought yesterday regarding things to consider when evaluating potential solutions:

Given usual real-time contexts, where resources are created/allocated at the start of the application (i.e., initialization phase), I've always mostly assumed that nodes/publishers/subscriptions/etc. are all created during the start of the application. See Sections III-B and IV-A in the [`ros2_tracing` paper](https://arxiv.org/pdf/2201.00393). This is usually true, but there are definitely cases where publishers/subscriptions are destroyed and new ones are created as time passes. For example, tools like RViz and the CLI tools (e.g., `ros2 topic pub`), or any node that dynamically subscribes to topics. It's a bit annoying if the trace data is incomplete in cases like that. Since we've had to start tracing from the beginning anyway, we've always correctly collected initialization trace data about any pub/sub created later, so this was never really a problem. However, it becomes more important now.

In summary, when making it possible to start tracing at any point in time, we should keep in mind that the objects (nodes/publishers/subscriptions/etc.) that are active when we want to start tracing might not necessarily have been active when the application was started. This means that collecting "initialization" information about the currently-active objects _at the time where we start tracing_ would be more optimal than collecting initialization information _at the start of the application_ (in case it is needed later). That is why LTTng has its [state dump mechanism](https://lttng.org/docs/v2.13/#doc-regenerate-statedump).

ShravanDeva5327, 2025-06-05:
> This is usually true, but there are definitely cases where publishers/subscriptions are destroyed and new ones are created as time passes. For example, tools like RViz and the CLI tools (e.g., ros2 topic pub), or any node that dynamically subscribes to topics.

I think we should consider labelling the tracepoints based on their types (initialization or runtime as in Table II of [ros2_tracing paper](https://arxiv.org/pdf/2201.00393)) and capture all the initialization data irrespective of it happening within the first few minutes after the application starts or at any point before the we start tracing. This way we don't have to bother about the objects that are active when we start tracing.

closed 2025-08-22 by christophebedard

christophebedard, 2025-09-09:
Just a follow-up to:

> This means that collecting "initialization" information about the currently-active objects _at the time where we start tracing_ would be more optimal than collecting initialization information _at [...]

@ShravanDeva5327 you looked into possibly doing a ROS 2 state dump like LTTng does. One of the problems was that it was hard to trigger the initialization tracepoints on demand at runtime (i.e., during the state dump). Another problem was that there was no real way to trigger the state dump like LTTng does for itself internally (i.e., when it starts recording).

To get around this, you looked into triggering initialization tracepoints at runtime alongside the runtime tracepoints, which was still a bit tricky and not very straightforward, not to mention that the initialization tracepoints got triggered multiple times.

I attended the [Tracing Summit 2025](https://tracingsummit.org/) last week. There was a [presentation about LTTng (slides are available on the talk page)](https://cfp.tracingsummit.org/ts2025/talk/QZPAVL/). One of the upcoming features that was mentioned was "application state dump callbacks," which is what we were trying to do/use here.

That still wouldn't enable this solution, because, again, it's hard to trigger initialization tracepoints at runtime. Also, this is more of a long-term feature part of [libside](https://github.com/efficios/libside), which isn't yet used by LTTng-UST. However, I thought I'd mention it here.

PR #191 (2025-08-22): Add support for starting tracing at runtime
## Description

At present, users must start tracing before launching their application in order to collect meaningful trace data, due to the way the ROS 2 source code is instrumented (See #190). 

This PR adds the ability to start tracing at runtime, provided some pre-configuration is done. Specifically, the `dual_session` argument of the `Trace` action must be set to `True`:

Setting the `dual_session` argument of Trace Action to `True`
```python
Trace(
    session_name='my-tracing-session',
    dual_session=True,
),
```
This starts an [LTTng sessi [... cut]