ros2/ros2_tracing #77: Colcon build error after installation ros2_tracing

yy945635407, 2023-08-24:
I am using Ubuntu 22.04 and ros2 humble.
I have finished installing ros2_traing by these commands:
```bash
sudo apt-get update
sudo apt-get install lttng-tools liblttng-ust-dev
sudo apt-get install python3-babeltrace python3-lttng

cd src/
git clone https://gitlab.com/ros-tracing/ros2_tracing.git
cd ../

colcon build --packages-up-to tracetools ros2trace
```
And the output in terminal of command `ros2 run tracetools status` is correct:
```bash
Tracing enabled
```

But when I tried to build a `rclcpp` project, I received an error message:
```bash
ylyt@ylyt-virtual-machine:~/ros2_ws$ colcon build --packages-up-to example_cpp --cmake-args -DTRACETOOLS_DISABLED=ON
Starting >>> example_cpp
--- stderr: example_cpp                             
CMake Warning:
  Manually-specified variables were not used by the project:

    TRACETOOLS_DISABLED

/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/node_01.dir/build.make:151: node_01] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/node_01.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< example_cpp [1.04s, exited with code 2]

Summary: 0 packages finished [1.24s]
  1 package failed: example_cpp
```

The code of the `cpp` file is as following:
```cpp
#include "rclcpp/rclcpp.hpp"

int main(int argc, char **argv)
{
    /* Initialize rclcpp  */
    rclcpp::init(argc, argv);
    /*produce a node named node_01*/
    auto node = std::make_shared<rclcpp::Node>("node_01");
   // Print string
    RCLCPP_INFO(node->get_logger(), "node_01 is started");
   
    rclcpp::spin(node);
    rclcpp::shutdown();
    return 0;
}
```

Similarly, when I tried to build `test_tracetools`, I got these errors:
```bash
ylyt@ylyt-virtual-machine:~/ros2_ws$ colcon build --packages-up-to test_tracetools
[0.440s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'tracetools' is in: /home/ylyt/ros2_ws/install/tracetools, /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding tracetools

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> tracetools_trace
Starting >>> tracetools_read
Starting >>> tracetools
Finished <<< tracetools [0.95s]                                                                                    
--- stderr: tracetools_read                                                         
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< tracetools_read [1.34s]
--- stderr: tracetools_trace
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< tracetools_trace [1.37s]
Starting >>> tracetools_launch
--- stderr: tracetools_launch                   
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< tracetools_launch [0.90s]
Starting >>> tracetools_test
--- stderr: tracetools_test                   
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< tracetools_test [0.91s]
Starting >>> test_tracetools
--- stderr: test_tracetools                             
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_timer.dir/build.make:151: test_timer] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:329: CMakeFiles/test_timer.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /opt/ros/humble/lib/librclcpp_lifecycle.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_lifecycle_node.dir/build.make:161: test_lifecycle_node] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:225: CMakeFiles/test_lifecycle_node.dir/all] Error 2
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_service_ping.dir/build.make:159: test_service_ping] Error 1
gmake[2]: *** [CMakeFiles/test_publisher.dir/build.make:159: test_publisher] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:355: CMakeFiles/test_service_ping.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:173: CMakeFiles/test_publisher.dir/all] Error 2
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_lifecycle_client.dir/build.make:159: test_lifecycle_client] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:251: CMakeFiles/test_lifecycle_client.dir/all] Error 2
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_ping.dir/build.make:159: test_ping] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:277: CMakeFiles/test_ping.dir/all] Error 2
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_pong.dir/build.make:159: test_pong] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:303: CMakeFiles/test_pong.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/test_intra.dir/build.make:159: test_intra] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:199: CMakeFiles/test_intra.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< test_tracetools [0.93s, exited with code 2]
                                
Summary: 5 packages finished [4.32s]
  1 package failed: test_tracetools
  5 packages had stderr output: test_tracetools tracetools_launch tracetools_read tracetools_test tracetools_trace
```

yy945635407, 2023-08-24:
Solved by cloning humble branch and redo:
```bash
git clone -b humble https://github.com/ros2/ros2_tracing.git
```

closed 2023-08-24 by yy945635407

christophebedard, 2023-08-24:
Yes, exactly. If you use ROS 2 Humble, then you need to use the `humble` branch.