ros2/ros2_tracing #78: rcl_init miss

Autostone-c, 2023-08-29:
[image]
hi, i meet this error? i saw the other didnt show any about this

christophebedard, 2023-08-29:
What version of ROS 2 are you using? What version of `tracetools_analysis`/`ros2trace_analysis` are you using?

christophebedard, 2023-10-13:
Since there has been no response, I'm going to close this. Feel free to re-open the issue and provide the versions of ROS 2/`tracetools_analysis` that you're using if you're still facing this issue.

closed 2023-10-13 by christophebedard

nightduck, 2024-05-14:
I'm getting the exact same error. Using the default binaries that come with Ubuntu 24.04

tracetools-analysis: 3.0.0-5noble.20240426.184931

I follow steps 1 and 2 [here](https://docs.ros.org/en/rolling/Tutorials/Advanced/ROS2-Tracing-Trace-and-Analyze.html#step-1-trace). After killing my process and running
    
    ros2 run tracetools_analysis process ~/.ros/tracing/test

I get the same `missing events: {'Ros2Handler': {'ros2:rcl_init'}}` error that @Autostone-c received.

The catch: on the same machine, with the same versions, I receive no error when I run a standard application from a public repo. The issue is definitely my code (which is definitely invoking rcl_init), but I can't find out what is wrong.

@Autostone-c : 
[image]

christophebedard, 2024-05-14:
> The catch: on the same machine, with the same versions, I receive no error when I run a standard application from a public repo. The issue is definitely my code (which is definitely invoking rcl_init), but I can't find out what is wrong.

Can you share the application you're tracing, or at least a minimal reproducer? Alternatively, you could use `gdb` to make sure that `rcl_init()` is called and that it runs the `TRACETOOLS_TRACEPOINT()` line. https://github.com/ros2/rcl/blob/6d53d24a863c3e9e4a41e9fe5f550271210d9d9d/rcl/src/rcl/init.c#L47

christophebedard, 2024-05-14:
Or alternatively, you could just comment out this line, assuming that you don't need this data: https://github.com/ros-tracing/tracetools_analysis/blob/760dca9598a04a64b2e1aef0f8e419e8018c79d6/tracetools_analysis/tracetools_analysis/processor/ros2.py#L107. Then it won't complain about it.

nightduck, 2024-05-14:
My application was directly linking to rcl code, specifically the rcl_timer_get_period function. When I moved this functionality down into a modified rclcpp layer, this issue disappeared.

So my guess is it's a linking issue. When you link directly to rcl, the tracer gets uppity.

nightduck, 2024-05-14:
Belay that. The issue came back. It does not seem to have anything to do with rcl_init in particular, because if I just execute the following simple script:

    from tracetools_analysis.loading import load_file
    from tracetools_analysis.processor import Processor
    from tracetools_analysis.processor.cpu_time import CpuTimeHandler
    from tracetools_analysis.processor.ros2 import Ros2Handler
    from tracetools_analysis.utils.cpu_time import CpuTimeDataModelUtil
    from tracetools_analysis.utils.ros2 import Ros2DataModelUtil
    
    # Load trace directory or converted trace file
    events = load_file('~/.ros/tracing/my_trace')
    
    print(events)

I get this printout:

    found converted file: /home/ubuntu/.ros/tracing/my_trace/converted
    []

nightduck, 2024-05-16:
For those who find this issue in the future, it can be caused by the following problems:

- The aforementioned rcl linking issue. Details unclear, but refrain from linking directly to rcl
- Mismatched root permissions between the tracer and the program. If the program is launched with root, the tracer must be launched with root
- Python environments. If your program's terminal has sourced a python environment that your tracer hasn't sourced, it will not work, (even if you prepend `env PATH="$PATH" LD_LIBRARY_PATH="$LD_LIBRARY_PATH"` to your command)
- Failing to prepend the `sudo env PATH="$PATH" LD_LIBRARY_PATH="$LD_LIBRARY_PATH"`. If you open a root terminal with `sudo su`, source ros and execute from there: also will not work.

The method I got working to execute a program in root is to put the following in a bash program:

    ros2 trace start test-trace
    ./path/to/my/program
    ros2 trace stop test-trace

And then execute that script with:

    sudo env PATH="$PATH" LD_LIBRARY_PATH="$LD_LIBRARY_PATH" bash run_and_trace_my_program.sh

christophebedard, 2024-05-16:
Just curious: why do you need to run as root?

nightduck, 2024-05-16:
> Just curious: why do you need to run as root?

My application uses real-time priorities

commit 846f77f9 (2020-03-18): Merge branch '78-launch_ros-get_default_launch_description-is-now-deprecated' into 'master'

Remove launch_ros.get_default_launch_description

Closes #78

See merge request micro-ROS/ros_tracing/ros2_tracing!160