ros2/ros2_tracing #86: lttngpy fails to build on Windows Debug

clalancette, 2024-01-11:
Looking at the nightly builds, it looks like the new `lttngpy` package fails to build on Windows Debug: https://ci.ros2.org/view/packaging/job/packaging_windows_debug/2070/

In particular, the end of the trace says:

```
--- output: lttngpy
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.20348.
-- The C compiler identification is MSVC 19.29.30153.0
-- The CXX compiler identification is MSVC 19.29.30153.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 2.3.2 (C:/ci/ws/install/share/ament_cmake/cmake)
-- Found Python3: C:/Python38/python.exe (found version "3.8.3") found components: Interpreter 
-- Found Python3: C:/Python38/python.exe (found version "3.8.3") found components: Interpreter Development Development.Module Development.Embed 
-- Found pybind11_vendor: 3.1.1 (C:/ci/ws/install/share/pybind11_vendor/cmake)
-- Performing Test HAS_MSVC_GL_LTCG
-- Performing Test HAS_MSVC_GL_LTCG - Success
-- Found pybind11: C:/ci/ws/install/opt/pybind11_vendor/include (found version "2.9.1")
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ci/ws/build/lttngpy
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System

  Building Custom Rule C:/ci/ws/src/ros2/ros2_tracing/lttngpy/CMakeLists.txt

  running egg_info

  creating lttngpy.egg-info

  writing lttngpy.egg-info\PKG-INFO

  writing dependency_links to lttngpy.egg-info\dependency_links.txt

  writing top-level names to lttngpy.egg-info\top_level.txt

  writing manifest file 'lttngpy.egg-info\SOURCES.txt'

  reading manifest file 'lttngpy.egg-info\SOURCES.txt'

  writing manifest file 'lttngpy.egg-info\SOURCES.txt'

  Building Custom Rule C:/ci/ws/src/ros2/ros2_tracing/lttngpy/CMakeLists.txt

  Building Custom Rule C:/ci/ws/src/ros2/ros2_tracing/lttngpy/CMakeLists.txt

  _lttngpy_pybind11.cpp

  status.cpp

LINK : fatal error LNK1104: cannot open file 'python38.lib' [C:\ci\ws\build\lttngpy\_lttngpy_pybind11.vcxproj]

---
Failed   <<< lttngpy [41.5s, exited with code 1]
```

@christophebedard can you take a look?

christophebedard, 2024-01-11:
> can you take a look?

Will do. I'm guessing this is something that other packages using `pybind11` have to deal with too, so I'll start there.

clalancette, 2024-01-11:
> Will do. I'm guessing this is something that other packages using `pybind11` (and thus Python) have to deal with too, so I'll start there.

Thanks, appreciated.

christophebedard, 2024-01-11:
See #87. Waiting for the test packaging_windows_debug job to finish.

PR #87 (2024-01-12): Fix Python not being found for lttngpy in Windows debug mode
Closes #86

I found this fix in a few other packages that use Python in CMake.
closed 2024-01-12 by clalancette