ros2/ros2_tracing #128: Request to bloom new release for Iron

Yadunund, 2024-07-11:
@christophebedard,

As part of the on-going preparation for the sixth patch release for Iron I tried to bloom `ros2_tracing` but was unsuccessful. I assume it's because I'm not in the [terraform group](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/tracing.tf).

Do you mind helping to bloom a new release? I have already updated the changelogs and pushed a new tag to the `iron` branch.
```bash
bloom-release --rosdistro iron ros2_tracing
```

christophebedard, 2024-07-11:
Here it is: https://github.com/ros/rosdistro/pull/42051

But that's weird, I believe other people have done releases before. Maybe we should add this repo to the ROS 2 core list here: https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/00-ros2_core.tf

Yadunund, 2024-07-11:
Thanks for helping bloom the release! 

@clalancette shall we add `ros2_tracing` to the core list?

clalancette, 2024-07-12:
> @clalancette shall we add `ros2_tracing` to the core list?

Yes please!  Please feel free to open a PR to https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/00-ros2_core.tf to add it.

Yadunund, 2024-07-12:
Done https://github.com/ros2-gbp/ros2-gbp-github-org/pull/560

I'll close this out!

closed 2024-07-12 by Yadunund