ros2/ros2_tracing #139: ros2 trace command is not recognized

YerayGG, 2024-10-07:
Dear Christophe Bedard,

I hope this message finds you well.

I am reaching out to report an issue I am experiencing with the message-link-instrumentation and humble branches of the ros2_tracing repository. Both branches were functioning properly for me at some point, but they have recently stopped working as expected.

I run ros2_tracing in a container, and I must reinstall it manually every time I launch the container. After installing all necessary packages, I checkout either the humble or the message-link-instrumentation branch, run colcon build, and source the workspace as usual. However, when I try to use ros2 trace, I encounter the following issue:

```bash
root@uwicore-pc5:/home/uwicore_pc5/TUM_BRIDGE/tracing_ws# colcon build --packages-up-to tracetools --allow-overriding tracetools
Starting >>> tracetools
Finished <<< tracetools [0.50s]                     

Summary: 1 package finished [1.74s]
root@uwicore-pc5:/home/uwicore_pc5/TUM_BRIDGE/tracing_ws# source ./install/setup.bash
root@uwicore-pc5:/home/uwicore_pc5/TUM_BRIDGE/tracing_ws# ros2 run tracetools status
Tracing enabled
root@uwicore-pc5:/home/uwicore_pc5/TUM_BRIDGE/tracing_ws# ros2 trace
usage: ros2 [-h] [--use-python-default-buffering]
            Call ros2 <command> -h for more detailed usage. ...
ros2: error: argument Call ros2 <command> -h for more detailed usage.: invalid choice: 'trace' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'test', 'topic', 'wtf')
root@uwicore-pc5:/home/uwicore_pc5/TUM_BRIDGE/tracing_ws# 
```

As you can see, after building and sourcing the workspace, the ros2 trace command is not recognized, even though the tracing status reports as enabled.

Could you please advise on how to resolve this issue? Any help would be greatly appreciated.

Thank you for your time and support.

Best regards,
Yeray

christophebedard, 2024-10-07:
The `ros2 trace` command is provided by the `ros2trace` package; it's separate from `tracetools` and the tracing instrumentation. Make sure you're also building the `ros2trace` package, or install `ros-$DISTRO-ros2trace`.

christophebedard, 2024-10-09:
I believe the above should resolve your issue, so I'm going to close this. If that doesn't help, let me know.

closed 2024-10-09 by christophebedard