ros2/ros2_tracing #149: Allow creating a live tracing session with `ros2 trace`/`Trace`

christophebedard, 2024-12-20:
When using the `lttng` CLI, this is done with the `--live` option: https://lttng.org/docs/v2.13/#doc-lttng-live.

It would be good to expose it through the `ros2 trace` command and the `Trace` launch action. A live session can be created using [`lttng-ctl`'s `lttng_create_session_live()`](https://github.com/lttng/lttng-tools/blob/c259edc03bedb9e0d58434cfcc27a7817772507a/include/lttng/session.h#L117-L131); this would need to be exposed through our `lttngpy`.

---

In the meantime, here's an example script to enable tracing for ROS 2 with a live session using the `lttng` CLI directly:

<details>

<summary>Click to expand: live_trace.sh</summary>

```sh
# live_trace.sh

if [ $# -eq 0 ]; then
  echo "Provide trace session name"
  exit 1
fi

session_name="$1"
hostname=$(hostname)

lttng create session_name --live
lttng enable-channel -u ros --subbuf-size=8M
lttng enable-event -c ros -u 'ros2:*'
lttng add-context -u -t vtid -t vpid -t procname
lttng start

echo
echo "Live trace URI: net://localhost/host/$hostname/$session_name"
echo "Press any key to stop & destroy"
read
lttng stop
lttng destroy
```

</details>

Here's a simple Python script that can receive and process some trace data live (just an example script):

<details>

<summary>Click to expand: bt2_lttng_live.py</summary>

```py
# bt2_lttng_live.py

import datetime
import sys
from typing import List

import bt2

def handle_trace_event(msg: bt2._EventMessageConst) -> None:
    event = msg.event
    ns_from_origin = msg.default_clock_snapshot.ns_from_origin
    timestamp = datetime.datetime.fromtimestamp(ns_from_origin / 1e9)
    print(f'{timestamp} {event.name}')
    # Accces field with event['field_name']
    # See:
    #   https://babeltrace.org/docs/v2.0/python/bt2/examples.html#get-a-specific-event-field-s-value
    # TODO
    if event.name == 'ros2:rclcpp_intra_publish':
        publisher_handle = event['publisher_handle']
        print(f'\tpub handle: {publisher_handle}')
    elif event.name == 'ros2:rclcpp_ring_buffer_dequeue':
        buffer = event['buffer']
        print(f'\tbuffer: {buffer}')

def process_live_trace(uri: str) -> None:
    # See example here:
    #   https://babeltrace.org/docs/v2.0/python/bt2/examples.html#create-explicit-source-components

    # Just switch source component to lttng-live:
    #   https://babeltrace.org/docs/v2.0/man7/babeltrace2-source.ctf.lttng-live.7/
    msg_it = bt2.TraceCollectionMessageIterator(
        bt2.ComponentSpec.from_named_plugin_and_component_class('ctf', 'lttng-live', {
            # See:
            #   https://babeltrace.org/docs/v2.0/man7/babeltrace2-source.ctf.lttng-live.7/#doc-_initialization_parameters
            'inputs': [uri],
        }),
    )

    # Ignore exception raised when we run out of events (i.e., consume too quickly)
    # TODO(christophebedard) not sure if there is a better way to do this
    while True:
        try:
            for msg in msg_it:
                if type(msg) is bt2._EventMessageConst:
                    handle_trace_event(msg)
        except bt2.TryAgain:
            pass

def main(argv: List[str]) -> int:
    if 1 != len(argv):
        print('provide 1 LTTng live path, e.g.: net://localhost/host/$HOSTNAME/session-name')
        print('print available session: babeltrace2 --input-format=lttng-live net://localhost')
        return 1
    live_uri = argv[0]
    print(f'LTTng live URI: {live_uri}')
    process_live_trace(live_uri)
    return 0

if __name__ == '__main__':
    sys.exit(main(sys.argv[1:]))
```

</details>

How to use:

1. `./live_trace.sh test`
    - This should print a URI starting with `net://`
    - The goal of this issue is to do this using `ros2 trace`/`Trace`, but for now this script can be used
1. `python3 bt2_lttng_live.py <URI>`
    - With the URI from step 1
    - This should be started before the application (step 3), otherwise [some important data may be missing depending on the analysis use-case](https://github.com/ros2/ros2_tracing#tracing)
1. Start your application

suchetanrs, 2025-03-24:
Hey @christophebedard !

I've been looking to contribute to this work since a while. I've gone through the issues related to this ticket.
I used your scripts to setup live tracing on a simple talker node. However, I received no outputs when I ran `babeltrace2 --input-format=lttng-live net://localhost/host/cluttered/my-trace-test`. Same with the python bindings. 

It seems like an issue with the ` --subbuf-size=8M`. When I don't specify a buffer size, all works well. I assume lttng is picking up a default size in the latter case. I am not sure about the root cause of this issue.

christophebedard, 2025-03-27:
Interesting, not sure why. Have you tried increasing the value, say, to 16M? If there are too many events, LTTng may have to write over some of them. Not sure if the live mode could have anything to do with it, but I can take a look tomorrow.

suchetanrs, 2025-03-28:
Thanks a lot for your reply! I restarted the container I was running and I have not been able to reproduce it since. Not sure what the issue was. It works for me with the 8M subbuffer value. 

I tried to create a live session using lttngpy. The PR for that is raised here https://github.com/ros2/ros2_tracing/pull/164.
The live session can be started using the below python file which mimics the `live_trace.sh` using `lttngpy` on the `rolling` branch.
<details>
<summary>trace_live_lttngpy.py</summary>
# trace_live_lttngpy.py

```py
#!/usr/bin/env python3
import os
import sys
import socket
from lttngpy import impl as lttngpy

def main():
    if len(sys.argv) < 2:
        print("Provide trace session name")
        sys.exit(1)
    session_name = sys.argv[1]
    hostname = socket.gethostname()

    # trace_dir = os.path.join("/root/lttng-traces", session_name)

    # Create a live session
    ret = lttngpy.lttng_create_session_live(session_name=session_name, url="net://localhost", timer_interval=1000000)
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to create live session '{session_name}': {error}")

    # Enable channel ros
    ret = lttngpy.enable_channel(
        session_name=session_name,
        domain_type=lttngpy.LTTNG_DOMAIN_UST,
        buffer_type=lttngpy.LTTNG_BUFFER_PER_UID,
        channel_name="ros",
        overwrite=0,
        subbuf_size=8 * 4096,
        num_subbuf=2,
        switch_timer_interval=200,
        read_timer_interval=200,
        output=lttngpy.LTTNG_EVENT_MMAP,
    )
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to enable channel 'ros': {error}")

    # enable events ros2:*
    ret = lttngpy.enable_events(
        session_name=session_name,
        domain_type=lttngpy.LTTNG_DOMAIN_UST,
        event_type=lttngpy.LTTNG_EVENT_TRACEPOINT,
        channel_name="ros",
        events={"ros2:*"},
    )
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to enable event 'ros2:*': {error}")

    # add more context fields to trace
    ret = lttngpy.add_contexts(
        session_name=session_name,
        domain_type=lttngpy.LTTNG_DOMAIN_UST,
        channel_name="ros",
        context_fields={"vtid", "vpid", "procname"},
    )
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to add context fields: {error}")

    # start tracing
    ret = lttngpy.lttng_start_tracing(session_name=session_name)
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to start tracing: {error}")

    print("")
    print(f"Live trace URI: net://localhost/host/{hostname}/{session_name}")
    input("Press Enter to stop & destroy...")

    # stop tracing
    ret = lttngpy.lttng_stop_tracing(session_name=session_name)
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to stop tracing: {error}")

    # destroy session
    ret = lttngpy.lttng_destroy_session(session_name=session_name)
    if ret < 0:
        error = lttngpy.lttng_strerror(ret)
        raise RuntimeError(f"failed to destroy session: {error}")

if __name__ == '__main__':
    if not lttngpy.is_available():
        print("LTTng control is not available.")
        sys.exit(1)
    main()
```
</details>

I will attempt to integrate the live tracing feature further with the tracing codebase. Please let me know if there are any issues with my PR or the python script :D

christophebedard, 2025-03-29:
> I restarted the container I was running and I have not been able to reproduce it since. Not sure what the issue was. It works for me with the 8M subbuffer value.

Weird! But good to hear that it works.

> I will attempt to integrate the live tracing feature further with the tracing codebase. Please let me know if there are any issues with my PR or the python script :D

I'll take a look at your PR.

suchetanrs, 2025-04-05:
Hello! @christophebedard 😄 

Thanks for sharing your branch. It gave me a good headstart and helped me understanding the code better!
I continued from where you left. Although it's still a WIP after my updates, I added a few things.
Description of what I added and the PR is here https://github.com/ros2/ros2_tracing/pull/165.

I modified the `--live` argument because figuring out if it was a live session through this made specifying the timer interval mandatory. I also added the option to specify the `url_origin` which defaults to `net://localhost`.

I am sure there is a cleaner way to do this instead of passing the `live_mode` parameter around everywhere but I am not sure what the best approach for that is.

suchetanrs, 2025-04-18:
Hi,

There seems to be a similar discussion here https://github.com/ros2/ros2_tracing/issues/157 about tracing specific nodes/topics/services etc. I am not entirely sure but I think the only way is to do this in post-processing after you know the pid/procname since the trace output contains that information. 

From what I've read, there's no direct solution to completely disable writing to the disk in live mode. However, the workaround is to limit the tracefile size for each channel and write to the tmpfs file system so that it only resides in memory. This can be done using the `--output` command when starting the LTTng relay daemon. More details here: https://lttng.org/man/8/lttng-relayd/v2.13/#doc-opt--output. A solution for a similar question is here: https://www.mail-archive.com/lttng-dev@lists.lttng.org/msg13246.html

There is another paper here https://www.landley.net/kdocs/ols/2006/ols2006v1-pages-209-224.pdf by the maintainer of LTTng which has some results about the overhead.

KARTOFF8xE, 2025-04-18:
Hi,

Like mentioned, we discussed a similar topic here #157.
What exactly do you mean with "trace callbacks"? If I understood you correctly, you ask if there is a way to trigger specific functions depending on which type of trace-message and e.g. topic-Name. You can do this by using BabelTrace2 :). You don't need to write to disk for that, but could process a single trace-Message right within your BabelTrace trace-Graph and trigger something right within there. Another possibility would be a communication from the trace-Graph to something else (e.g. using MQTT) and trigger a more complex function there.

Sorry, I hope this message isn't too messy.

ashwinvkNV, 2025-04-18:
Thanks for the responses.

Disabling Disk Writing (To reduce file writing overhead when only live measurements are needed):
1. @suchetanrs , thanks for the suggestion. Writing to tmpfs should indeed be faster, but to find out by how much, I’ll need to measure the overhead using the [ros2_tracing-analysis-example](https://github.com/christophebedard/ros2_tracing-analysis-example).
2. @KARTOFF8xE , you mentioned "You don't need to write to disk for that.". Yes, I was able to directly process the "trace message" directly within BabelTrace using the example at the start of this issue. However, based on the [LTTng live documentation](https://lttng.org/docs/v2.13/#doc-lttng-live), it seems like disk writing is still happening. Specifically, it says: "allow compatible trace readers to display or analyze event records as LTTng records events on the target system **while recording is active**." Or do you know if "recording" only means triggering the trace events and not writing to disk?

Trace Filtering (To reduce overhead by enabling only relevant trace events):
1. @KARTOFF8xE, By "trace callbacks", I was referring to `handle_trace_event`. It seems like we can filter events by event type (e.g., lttng enable-event --userspace 'ros2:rcl_publisher_init'). However, it doesn’t seem possible to enable/disable events based on attributes like topic name(since this is only available after the event is triggered). We can filter all 'ros2:rcl_publisher_init` events and extract information for a specific topic, but I suspect this could still introduce significant overhead (unless file writing is disabled). For example, this could be an issue when there are many publishers, but users are only interested in tracing events and statistics for a subset of them.

christophebedard, 2025-04-22:
LTTng is commonly used in production (e.g., networking, telecom), even on fairly low-power (CPU/RAM) systems. In those cases -- especially for systems with no real storage -- the trace data is sent over the network to a remote system that records it.

I believe you're all referring to the relay daemon writing to disk, but you can see in the section just above the "View events as LTTng records them" section that the relay daemon is started on the remote system when sending trace data to a remote system: https://lttng.org/docs/v2.13/#doc-sending-trace-data-over-the-network.

Obviously, if you're tracing and doing live trace data processing on the same system, then you do indeed need the relay daemon on that system.

> Specifically, it says: "allow compatible trace readers to display or analyze event records as LTTng records events on the target system while recording is active." Or do you know if "recording" only means triggering the trace events and not writing to disk?

Yeah, I believe "recording" simply generally refers to trace data being generated when a tracepoint is hit. It doesn't necessarily imply that it is written to disk on the system that's tracing.

> Additionally, monitoring every topic or interface might not be required. Are there any efforts or existing mechanisms to selectively enable or disable tracing based on trace types or the names of interfaces (e.g., topics, services, actions)?

This is definitely an interesting question! There have been similar questions in the past, as mentioned above. However, as you figured out, LTTng has no concept of nodes or topics (and while it can [filter events based on event data/payload](https://lttng.org/man/1/lttng-enable-event/v2.13/#doc-filter-cond), you couldn't filter `ros2:*_publish` events based on topic names, because it's not in the event payload), so it can't help here. You could try to segment your application by process and only trace certain processes, but you'd need to know the PIDs before tracing, and you'd need to start tracing before launching your application, so it would require a bit of coordination.

This is definitely something I'd like to explore, but I unfortunately haven't done so yet. However, with #44 and this issue, we're moving in that direction.

Let us know if you have other questions.

TanayK07, 2026-10-21:
@ashwinvkNV late reply, but i ended up building something for the same use case and it might help.

ros2_pulse is an LD_PRELOAD shim over the tracetools hooks instead of an lttng session, so nothing gets written per event. it counts per topic in process and writes one small summary every few seconds. no sessiond or relayd, nothing on disk except that summary.

on a ~4900 msg/s stress graph it cost about +1.9% cpu pooled across humble, jazzy and kilted, roughly 1 µs per message. since 0.7.0 pulse_bridge turns expected rate rules into /diagnostics statuses, which sounds like what you were after. filtering is in the rules file, so it still counts every topic but only reports the ones you list.

it won't replace live tracing for latency or callback level stuff, only rates and gaps. docs: https://docs.ros.org/en/jazzy/p/ros2_pulse/