ros2/ros2_tracing #150: Will there be new Tracepoints for older versions

KARTOFF8xE, 2025-01-20:
Hi,

I noticed that some information exists only for later versions and are listed as ToDo in older ones.

I.e. [Client request/response](https://github.com/ros2/ros2_tracing/blob/rolling/doc/design_ros_2.md#client-requestresponse) exists for rolling but not for humble.

Will these be brought to older versions someday?

christophebedard, 2025-01-22:
We've never backported new tracepoints to older distros, and there is currently no plan to do so.

It can be discussed, but new "features" are typically not backported to older distros.

KARTOFF8xE, 2025-01-22:
In this case I would just like to have a question:
Am I right in assuming that the [Client request/response](https://github.com/ros2/ros2_tracing/blob/rolling/doc/design_ros_2.md#client-requestresponse) is the only way to bring client and server into relation by only looking onto the Trace-Data?

If yes: could you please give me a short hands-on on how to implement it myself? Or wouldn't you recommend that?

christophebedard, 2025-01-23:
> In this case I would just like to have a question:
Am I right in assuming that the [Client request/response](https://github.com/ros2/ros2_tracing/blob/rolling/doc/design_ros_2.md#client-requestresponse) is the only way to bring client and server into relation by only looking onto the Trace-Data?

I'm not sure I really understand your question. It is _one_ solution for tracking/linking service requests & responses using trace data, yes. If you wanted to achieve the same thing, you would have to basically implement the same thing for Humble. In practice, this means backporting a list of PRs, including #145 and all pull requests linked in the description (or just backport the PR for the `rmw` implementation you're using, which would be https://github.com/ros2/rmw_fastrtps/pull/787 if you're using Fast DDS).

Since Humble is "old" at this point, to minimize the potential additional headaches caused by backporting these PRs to Humble (you might get merge conflicts when backporting recent code to an older distro, or might have to backport other pull requests), I would recommend switching to a newer distro if at all possible.

KARTOFF8xE, 2025-01-27:
Wait, is there another way to track service requests&responses using _ros2_tracing_ I overlooked? Either case, which way would you recommend (for humble)?

Feel free to close the Issue with your answer.

Thanks for the quick reply and the answers :).

christophebedard, 2025-01-27:
> Wait, is there another way to track service requests&responses using ros2_tracing I overlooked?

There is not. I just meant that theoretically, there could be many solutions, and #145 is just _one_ of the solutions. However, there is no other way currently.

> Either case, which way would you recommend (for humble)?

If you really want to do it with Humble, try to backport the list of PRs I provided above. You'll have to build ROS 2 from source (or at least the parts of ROS 2 you'll have to modify, like `tracetools` and `rmw_*`).

closed 2025-01-27 by christophebedard