ros2/ros2_tracing #155: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'

edu-rossrobotics, 2025-03-10:
Hi, I'm trying to build a bunch of packages with tracing enabled. On most of them it builds without any issue but on some of them I get this error:

```bash
--- stderr: imu_driver
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `tracetools::detail::get_symbol_funcptr(void*)'
```

I'm building it under ROS2 humble.

christophebedard, 2025-03-10:
I would need a bit more information to be able to help:

1. I'm guessing you installed ROS 2 Humble from apt?
2. I'm guessing you're just building your own packages and _not_ building any of ROS 2 from source, except for `tracetools`?
3. How are you building your packages with tracing enabled? Please provide the exact `colcon build ...` command.
4. What exact dependency are you adding to a `CMakeLists.txt` and to which package are you adding it to? It would probably be helpful to know what dependencies this package has.

edu-rossrobotics, 2025-03-10:
1. Yes, ROS 2 has been installed from apt. Actually I'm trying to build within a docker container where the base image is [this](https://hub.docker.com/layers/arm64v8/ros/humble/images/sha256-3ca05128afeb4e4f0d38f0f070825224dcee2c0267787324cc83a514eec9cb66). Note that this is running on an ARM device, in case it can make any difference.

2. Yes, I've just compiled `tracetools` according to the instructions on README file for Humble branch.

3. The exact command is the following:

```
colcon build \
    --merge-install \
    --symlink-install \
    --parallel-workers 4 \
    --cmake-args "-DCMAKE_BUILD_TYPE=$BUILD_TYPE" "-DCMAKE_EXPORT_COMPILE_COMMANDS=On" \
    -Wall -Wextra -Wpedantic
```

4. I've added `tracetools`, so the whole dependencies on the `CMakeLists.txt` are the following:

```
find_package(ament_cmake REQUIRED)
find_package(ament_cmake_python REQUIRED)
find_package(rclpy REQUIRED)
find_package(ament_cmake_ros REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rclcpp_components REQUIRED)
find_package(diagnostic_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(yaml-cpp REQUIRED)
find_package(std_srvs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(std_msgs REQUIRED)
find_package(mk4_interfaces REQUIRED)
find_package(tracetools REQUIRED)
```

So adding the dependency fixed the issue for that specific package but still appears on other packages. If there a way to get more information from the error I'm happy to test it but I couldn't figure out how to debug it more.

Thnaks.

edu-rossrobotics, 2025-03-11:
I've just realize I was using rolling branch on humble distro so evidently it wasn't working as expected. Switched back to humble branch and all the issues disappear. Sorry for the confusion.

closed 2025-03-11 by edu-rossrobotics

christophebedard, 2025-03-11:
Sorry, I haven't been able to get to it yet, glad you were able to figure it out. But, yes, that makes sense! This is not uncommon, which is why we have this in the README 😁:

> **Note**: make sure to use the right branch, depending on the ROS 2 distro: [use rolling for Rolling, galactic for Galactic, etc.](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Developer-Guide.html#branches)