ros2/ros2_tracing #157: Question: Tracing specific node

mamariomiamo, 2025-03-14:
Hi, may I check if there is a way to specify which node we trace?

KARTOFF8xE, 2025-03-14:
Hi, do you only want to track a specific node or process or do you want to know how to differentiate between nodes or processes to know which node or process you are tracing?

mamariomiamo, 2025-03-14:
I am wondering if I could just track specific node/process.

KARTOFF8xE, 2025-03-14:
As far as I know (but I'm not a dev here) this is not possible. Tracepoints can only be enabled or disabled. But you can't make restrictions in terms of what Nodes/Subs/Pubs/Services/etc. you want to trace.

KARTOFF8xE, 2025-03-14:
But what you can do is using the [Babeltrace API](https://babeltrace.org/docs/v2.0/libbabeltrace2/). Meaning using the [source.ctf.fs](https://babeltrace.org/docs/v2.1/man7/babeltrace2-source.ctf.fs.7/) plugin for files or [source.ctf.lttng-live(7)](https://babeltrace.org/docs/v2.1/man7/babeltrace2-source.ctf.lttng-live.7/) for live-sessions. For latter one you can not use ros2_tracing directly but need to use lttng (I can give you some commands if you want to, but lttng also has an API if you want to do it programmatically). Following you write a filter or sink component using the Babeltrace API. Babeltrace provides usefull examples.

christophebedard, 2025-03-14:
You indeed can't decide to trace only specific nodes, but you can use get LTTng to only trace certain processes with the `lttng enable-event --filter` option, see https://lttng.org/docs/v2.13/#doc-enabling-disabling-events. You'd have to configure tracing using the `lttng` CLI instead of `ros2 trace`. However, this doesn't really help in practice, because you [need to configure/start tracing before launching your application](https://github.com/ros2/ros2_tracing#tracing), so you don't know the PID at that time. And you can't get LTTng to only trace certain nodes, because it has no concept of ROS 2.

The simplest solution is to filter the trace data yourself, like @KARTOFF8xE said. You could also just trace as usual and write a babeltrace2 plugin to filter the trace data later/offline.

If you can share _why_ you want to only trace specific node or process, I could provide more options.

mamariomiamo, 2025-03-18:
The main reason to only trace specific node/process is because normally the trace log is quite big if it run for an extended duration and thus I was thinking one way to limit the size is not to trace all but some nodes of interest.

KARTOFF8xE, 2025-03-18:
Tbh, i got a similar problem. So I'm pretty interested how @christophebedard would fixe it. Adapted to your problem I would recommend doing it this way:

First instead of using ros2_tracing use lttng directly, do something like the following Script:

``` bash
#!/bin/bash

lttng-relayd -d
lttng create tracingSession --live 1000000 -U net://localhost
lttng enable-event --userspace 'ros2:rcl_node_init'
lttng enable-event --userspace 'ros2:rcl_publisher_init'
lttng enable-event --userspace 'ros2:rcl_subscription_init'
lttng enable-event --userspace 'ros2:rcl_service_init'
lttng enable-event --userspace 'ros2:rcl_client_init'

lttng add-context --userspace --type=procname
lttng add-context --userspace --type=vpid 
lttng start
```

It enables some Tracepoints and sends their messages to an endpoint.

Next you write your own tracegraph using the BabelTrace [C-Api](https://babeltrace.org/docs/v2.1/libbabeltrace2/). Fortunatelly, this should be pretty easy for your usecase:

You need one source, filter and sink Component each.

The Source should be the following (worked perfectly fine for me): [source.ctf.lttng-live](https://babeltrace.org/docs/v2.0/man7/babeltrace2-source.ctf.lttng-live.7/)

You need to configure its arguments (well documented on the linked website).

The filter component:
You need to write this one by your own. But there is a great [example](https://babeltrace.org/docs/v2.1/libbabeltrace2/example-simple-flt-cmp-cls.html)-usecase. Just start from there.

As sink you could use:
[sink.ctf.fs](https://babeltrace.org/docs/v2.0/man7/babeltrace2-sink.ctf.fs.7/)

Configure its arguments.

Now you connect their ports:
The output-port from the source goes into the input of the filter. Analogue for filter and sink.

christophebedard, 2025-03-28:
> The main reason to only trace specific node/process is because normally the trace log is quite big if it run for an extended duration and thus I was thinking one way to limit the size is not to trace all but some nodes of interest.

You could also limit the tracepoints that are enabled. Depending on your analysis use-case, you can remove a number of them.