ros2/ros2_tracing #178: Fix clang/gcc warnings

christophebedard, 2025-05-21:
### Operating System:

Ubuntu

### ROS version or commit hash:

Rolling (latest)

### Steps to reproduce issue

1. Build `tracetools` using the current `rolling` branch with clang and libc++:
    ```
    $ colcon build --packages-up-to tracetools --mixin clang-libcxx --cmake-args -DCMAKE_BUILD_TYPE=Debug
    ```
    * See the build command in the CI job here: https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/1860/consoleFull#console-section-12.
        * The latest nightly_linux_clang_libcxx job is: https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/2260/. You can see here that there are still warnings in `tracetools`: https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/2260/clang/
    * This section in the ROS 2 docs also explains how to build with clang, but it doesn't cover building with libc++: https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html#clang.
1. Check the build output for `tracetools` and see if warnings are reported

### Expected behavior

No issues/warnings

### Actual behavior

There are some issues reported by clang, see https://github.com/ros2/ros2_tracing/pull/115.

Some of those issues may have been fixed since that PR was opened over a year ago.

### Additional information

After confirming that there are warnings, try to fix them. Look at the changes in #115.

After fixing the warnings, re-build and confirm that the warnings disappeared.

christophebedard, 2025-05-21:
@ShravanDeva5327 I created this issue and included instructions on how to build with clang so that you can reproduce the build warnings and try to fix them.

Let me know if you have any questions.

ShravanDeva5327, 2025-05-22:
@christophebedard Thankyou for these instructions, I'll start looking into it.